klouderone/CorneV4ProMicroEdition #1: Choc v2 Compatibility

Gustavo-DMS, 2024-06-25:
The Corne V4 by foostan is already compatible with Choc v2 switches, is there any reason why this version isn't? And maybe put a warning on the README.

klouderone, 2024-06-26:
Will update to include choc v2 support. I didn't originally add them as V2 choc switches aren't as common as V1 switches.

Gustavo-DMS, 2024-06-26:
I does make a lot of sense, and I think it's more interesting to have a single PCB compatible with MX and choc switches.

Choc v2 requires a 5mm hole for it's stem versus MX's 4mm, if I cut the extra pin does the board have clearance for this extra millimeter?

BTW. I'm building two boards with a nice!nano, one with chocs and another with mx, feel free to ask any questions.

klouderone, 2024-06-28:
Unfortunately no, the centre hole isn't 5mm in diameter if I remember correctly, but it should be possible to increase that hole to 5mm diameter so Choc V2 without the top pin is compatible. I will do this when I get back from holiday on the 12th July.

closed 2024-08-12 by klouderone