klouderone/CorneV4ProMicroEdition #3: improved Choc switch plates

LeoLivet, 2024-09-22:
I am close to finishing my wireless LP build with this but am having issues with the swithplate being loose. It seems the included one was mostly meant for MX style switches. By being 1.40mm thick (measured with calibers (i know realise it might also have been my slicer settings too))

I have tried to modify the switch plate after finding this [reddit thread about choc switchplates](https://old.reddit.com/r/ErgoMechKeyboards/comments/pl3xwt/kailh_choc_v1_plate_question/) where /u/kbjunky links a picture of a 2.2mm switch plate, but with an indent where the choc switches can slot into a plate of 1.32 mm. 

I have sort of succesfully made a modified switch plate that has the indents by projecting the stil lines but i think with my limited skill in fusion it might be a little inaccurate when exporting to the slicer.

I think there also might be a little room to decrease the tightness of the the holes for a little tighter fit now, the switches are a tiny tiny bit loose still.

These plates were printed with Bambu P1S, with bambu PLA galaxy.

Also i didnt have the correct spacers so 4 mm ones but it now looks like there is room for lowering the sides atleast 3mm, but havent come to that point yet.
[image]
[image]
[image]

[New corne v4 pro micro choc switchplate test right.zip](https://github.com/user-attachments/files/17089070/New.corne.v4.pro.micro.choc.switchplate.test.right.zip)

klouderone, 2024-09-23:
hey mate, i have noticed this too. i shall print these for my own board to test and then add to the github repo! thanks!

klouderone, 2024-09-23:
printed it, works perfectly for me, only a touch of play up and down, no play side to side. could you please send me the STEP file? I shall modify for 3x5 layout too and then add to repo and give credit. cheers!

LeoLivet, 2024-09-23:
[Corne V4 Pro Micro Edition 6 Column Switch Plate v4 modified thickness with slots for choc.zip](https://github.com/user-attachments/files/17095501/Corne.V4.Pro.Micro.Edition.6.Column.Switch.Plate.v4.modified.thickness.with.slots.for.choc.zip)
I attached it here in a zip file. I was mostly worried about the play because i was just using the stp file you provided. And fusion was doing some things that made the holes look weird.

But basically i just projected two different plates.

One with 0.9mm thickness and then another with 1.3mm thickness 
Then i just offset the 0.9mm faces with -1 mm for the holes.

i did a few test prints of some different thickness and it seems in the end to get the desired thickness sort of comes to down to the resolution you use on the z-height in the slicer. i could seeming only get it to be between 1.28 or 1.32 according to the slicer. :)

klouderone, 2024-09-23:
i realised i could just model it myself after i sent that message haha. With the desired thickness, depending on the slicer you use, there is an option called adaptive Z height, which allows you to get pretty close to the modelled thickness using different layer heights. Shall print today!

klouderone, 2024-09-26:
Printed and works very well, shall be adding. Thanks for bringing this to my attention!

LeoLivet, 2024-09-27:
nice! i am thinking about trying to make another modification to the case itself to make even thinner, i dont know if its bad by convention to have the hotswap sockets touch the bottom. But i think the height can be lowered by another few mm too. 
(perhaps if you have the step files with timeline for fusion and wanna share it that'd be cool? :) )
Anyways i will mark this as closed if that's the correct way to do so. :)

closed 2024-09-27 by LeoLivet

klouderone, 2024-09-28:
i don't use fusion unfortunately, i use Rhino. the step files are there if you want to use them to make your own model though! it is also fine to have them touch the bottom, plastic isn't conductive!