klouderone/CorneV4ProMicroEdition #6: about firmware

charleyzhu, 2024-11-19:
Hello, I am currently using the RP2040 Pro Micro module. After uploading the firmware to the left keyboard, it always stays in the bootloader and cannot load the firmware properly. The right keyboard's LED is displaying content. What could be the issue? Also, is this firmware open source?

klouderone, 2024-11-19:
Firmware is open source, and is under Corne on ZMK, QMK and QMK-VIAL. The precompiled firmware is proven to work, and is added as a quick check to make sure everything works as intended. Sounds like a microcontroller problem?

klouderone, 2024-11-20:
How are you writing to the RP2040? Are you entering the bootloader by double clicking the reset button, and then dragging the UF2 file into the controller on your file explorer? I am not sure what the problem would be other than the microcontroller or the correct flashing method.

charleyzhu, 2024-11-21:
> How are you writing to the RP2040? Are you entering the bootloader by double clicking the reset button, and then dragging the UF2 file into the controller on your file explorer? I am not sure what the problem would be other than the microcontroller or the correct flashing method.

I’m sure I’m writing the firmware correctly, and I’m also considering the controller issue. I’ve purchased two RP2040 boards with 16M Flash (the current Flash size is 4M). I’ll try writing the firmware again once they arrive tomorrow.

closed 2024-12-15 by klouderone