klouderone/CorneV4ProMicroEdition #9: Battery Support

rodridevops, 2024-12-20:
Thank you for this beautiful version of Corne!

We are working on a personal project and greatly miss the microcontroller support from the 4th version.

Could you please share your timeline for adding support for an external battery connector to the PCB (instead of the Pro Micro)? Also, are you considering adding an on/off switch to the design?

klouderone, 2024-12-20:
Hey mate, 

There is currently a switch on the PCB for battery control, as well as battery pads. Only thing missing is a battery connector. Still got to get around to do it, though I don't have too much time at the moment to do it with work. I'll see what I can come up with this weekend, if I get a moment I'll get around to doing it.

Will update this issue when I update the repo!

rodridevops, 2024-12-20:
Thank you so much Claude. I will test it out as it is today!

I will let you know here and on reddit :)

ffiebig, 2024-12-22:
Thanks for the awesome work on this project @klauderone, excited for the upcoming update

klouderone, 2024-12-22:
Haven't managed to get around to it this weekend, family turned up a day early for Christmas, will be update Boxing Day!

klouderone, 2025-02-16:
Hey there all, sorry i have been absent and didn't get around to doing this earlier. I had some great opportunities come up in my personal life that put this on hold. Rev 2 is now live, with right angle battery connector support, and through hole diode support. Sorry for the delay!

closed 2025-02-16 by klouderone