klouderone/CorneV4ProMicroEdition #10: Requesting symbols!

thomaseliot, 2025-01-17:
Hi there! I'm wondering if you would be willing to push the footprints and symbols for the switches? Such as `ClaudeFoot:SW_MXchoc_HS_1u` I would love to use this design as a base for another one.

thomaseliot, 2025-01-17:
I'm also trying to see if I can get the board populated by JLPCB, I think those would help!

klouderone, 2025-02-16:
Hey mate, please check my github repositiories for my Kicad footprint library! Unfortunately i dont have any symbols, to give you as i never made any, just used other ones built in or found in other places. To be completely honest, my kicad library is a mess, and i am only just realising it as I never spent the time to actually organise it or find a way to make it easily shareable when i first started. Hope this can help. Also sorry for the late reply, i had some things come up!

closed 2025-02-16 by klouderone