klouderone/CorneV4ProMicroEdition #14: OLED Cover and TPU Sound Dampener

SparrowMike, 2025-07-06:
Hi!

Seems like the OLED Cover Left.stp is corupted, I'm getting a message of "The file does not contain any geometry data." and "Meshing of a model file failed or no valid shape." inside the bambu studio.

Also any chance for a 6 column TPU Sound Dampener?

Many thanks!

Cheers

klouderone, 2025-07-09:
Will take a look into both of these when I get a minute later today. Thanks for bringing to my attention!

closed 2025-07-11 by klouderone

SparrowMike, 2025-08-17:
Hi! It seems like the nice view cover left is the same as no display 

[image]

klouderone, 2025-08-17:
Hm. Will investigate and update.